ricosjp/ruststep #54: Case sensitivity in EXPRESS schema

termoshtt, 2021-05-05:
As far as I read ISO-10303-11, user defined identifiers should be snake_case, but some identifier copied into `schema/` does contains non snake_case identifier like `degree_Celsius`

g-rauhoeft, 2021-05-12:
I believe I've found the cause of the test failures if mixed case is allowed:

For blocks without an END_ tag, such as UNIQUE, the parser swallows the END_ tag of the parent block, because it doesn't know how to differentiate between the end of the UNIQUE block and its contents. For this reason I suggest creating either an `until_tag` combinator, that consumes the input until an END_ tag is reached, or an `is_not_reserved` combinator, that wraps a  `nom::combinator::cond combinator`, checking whether an identifier is actually a keyword reserved by EXPRESS.

I'm currently working on the latter and hope to make a pull request early next week.

g-rauhoeft, 2021-05-21:
I've started implementing the changes under: https://github.com/g-rauhoeft/ruststep/tree/case-sensitivity-fix

Please let me know if you feel this is going in the right direction. I haven't created a pull request yet, because some test cases still fail, but a vast majority are successful.

termoshtt, 2021-09-27:
https://github.com/ricosjp/ruststep/pull/66#issuecomment-868471563
>  6.1 The syntax of the specification refers to keywords being in uppercase, but that is only for the syntax of the specification in WSN, not for EXPRESS as such.

This looks true.

> As such, restricting any identifiers to lower case goes against the specification.

I agree. We should accept non-lower case identifiers.

> or an is_not_reserved combinator,

Detecting reserved keyword in `simple_id` has been implemented in #87, and improved in #117

> For this reason I suggest creating either an until_tag combinator, that consumes the input until an END_ tag is reached,

The `many_till` parser in #66 has been merged though #114, but not used in esprc.