ricosjp/ruststep #55: espr::parse fails for input `ENTITY\n`

termoshtt, 2021-05-05:


g-rauhoeft, 2021-06-07:
I've made progress in fixing TYPE parsing. For me the line ```expression_is_constant(SELF);``` failed to parse. The reason for this is because the ```function_call``` parser first recognized the built-in express ```exp``` function and then tried to parse the remainder, which was ```ression_is_constant(SELF);```, resulting in a syntax error. Fixing this requires the use of the work that I submitted earlier regarding reserved keywords. I'll create a new pull request within the next couple of days containing a fix for this.

g-rauhoeft, 2021-06-08:
The work I've put in to this is located in https://github.com/g-rauhoeft/ruststep/tree/entity_fix_55.
Entities are parsed fine as far as I can see, but it fails on certain functions. It must be due to more order of operation problems, which I'll sort out. Should I make a pull request for this anyway, or would you like to wait until I've fixed the remaining issues?

termoshtt, 2021-09-27:
Fixed in #66

closed 2021-09-27 by termoshtt