ricosjp/ruststep #157: Does the transpiler support IFC schemas?

ckabanur, 2021-11-18:
I tried to use esprc transpiler for IFC schema (ver 2x3) [https://standards.buildingsmart.org/IFC/RELEASE/IFC2x3/FINAL/EXPRESS/IFC2X3_Final.exp](url) as input, using the below terminal command:

`./target/release/esprc ./schemas/ifc/IFC2x3.exp | rustfmt > ./ruststep/src/ifc2x3.rs`

It results in following error:

`Syntax Error occurred while parsing following line [Nom(Tag)]:
> FUNCTION IfcCorrectObjectAssignment
>       (Constraint: IfcObjectTypeEnum; Objects : SET[1:?] OF IfcObjectDefinition)
>       : LOGICAL;
>     
>   LOCAL
>     Count : INTEGER := 0;
>   END_LOCAL;
>   
>       IF NOT(EXISTS(Constraint)) THEN 
>         RETURN(TRUE);

Syntax Error occurred while parsing following line [Nom(Many1)]:
> SCHEMA IFC2X3;
> 
> TYPE IfcAbsorbedDoseMeasure = REAL;
> END_TYPE;
> 
> TYPE IfcAccelerationMeasure = REAL;
> END_TYPE;
> 
> TYPE IfcAmountOfSubstanceMeasure = REAL;
> END_TYPE;

thread 'main' panicked at 'Syntax Error', espr/src/bin/esprc.rs:35:13`

If I increase the number of error lines in the command (like 10000), I see that it basically dumps the entire content of the schema file as syntax error ([Nom(Tag)] and [Nom(Many1)].

The code is too complex for me to debug and understand. Please advise.
Does esprc support IFC schemas? Is it planned for later releases?

termoshtt, 2021-11-18:
Currently (0.1.0 release), esprc can only understand a limited grammar of EXPRESS, and tested only for AP201 schema. We are still working to support entire EXPRESS language until 1.0 release.

> thread 'main' panicked at 'Syntax Error', espr/src/bin/esprc.rs:35:13`

"Syntax Error" means esprc fails to tokenize input. A better way I usually do for debugging espr is to create a minimal input to cause syntax error.

ckabanur, 2021-11-18:
Thanks for the clarification and guidance!